xpack-dev-tools/qemu-arm-xpack #3: stm32f407 uart input/output

darleep, 2020-05-03:
### Prerequisites

#### Check the support page

In case you encounter problems while using the **xPack QEMU Arm**, please
read carefully the [support](https://xpack.github.io/qemu-arm/support/) page.

#### Discuss the problem in the forums

If you still could not find a solution, if you have interesting use cases,
if you have custom configurations, and generally if you have any
experience that you want to share with others,
use the [xPacks forums](https://www.tapatalk.com/groups/xpack/).

If the forum discussions concluded that you identified a bug, or you have a
pertinent suggestion how to enhance xPack QEMU Arm, continue and register
an issue.

> Note: This project is dedicated to building the xPack QEMU Arm and only
issues related to these topics should be registered here; for functional
issues please use the separate
[xpack-dev-tools/qemu](https://github.com/xpack-dev-tools/qemu/)
project.

### Description

[Description of the bug or feature]

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behaviour:** [What you expected to happen]

**Actual behaviour:** [What actually happened]

### Versions

* QEMU package version
* operating system and version
* toolchain version
* the **detailed and complete list of steps to reproduce the bug** (mandatory)

Please understand that without being able to reproduce the bug we cannot
identify your problem.

To be allowed to enter issues, you need first to login to GitHub.

Note: the text should respect the markdown syntax; preview the message
before posting and correct if it does not look as expected, especially
inline code or other quoted text.

---

Note: Remove unnecessary text after reading, before entering your ticket.

ilg-ul, 2020-05-03:
Sorry to ask, but can you read English?

darleep, 2020-05-09:
Yes i can

ilg-ul, 2020-05-09:
Then please read your bug report.

And do not post to this tracker via email, use the web interface; or at least remove the quoted text from the message.

closed 2020-06-09 by ilg-ul