xpack-dev-tools/qemu-arm-xpack #4: How to add devices

LuoZhongYao, 2020-07-24:
The default board only provides a few devices for use, for example STM32F107VC only provides
```
'/machine/mcu/stm32/RCC', address: 0x40021000, size: 0x0400
'/machine/mcu/stm32/FLASH', address: 0x40022000, size: 0x0400
'/machine/mcu/stm32/PWR', address: 0x40007000, size: 0x0400
'/machine/mcu/stm32/AFIO', address: 0x40010000, size: 0x0400
'/machine/mcu/stm32/EXTI', address: 0x40010400, size: 0x0400
'/machine/mcu/stm32/GPIOA', address: 0x40010800, size: 0x0400
'/machine/mcu/stm32/GPIOB', address: 0x40010C00, size: 0x0400
'/machine/mcu/stm32/GPIOC', address: 0x40011000, size: 0x0400
'/machine/mcu/stm32/GPIOD', address: 0x40011400, size: 0x0400
'/machine/mcu/stm32/GPIOE', address: 0x40011800, size: 0x0400
GDB Server listening on: 'tcp::1234'...

```
How to add more devices, such as USART, ADC, etc.
I tried -serial and -chardev to no avail

ilg-ul, 2020-07-24:
QEMU does not allow to dynamically add devices, they need to be compiled into the binary.

LuoZhongYao, 2020-07-24:
> 
> 
> QEMU does not allow to dynamically add devices, they need to be compiled into the binary.

I see that there are USART devices in STM32F107xx-qemu.json, is there a way to use these devices?

ilg-ul, 2020-07-24:
The USART devices were contributed by someone else, they might work with some restrictions, but I never used them.

If you want to simply display trace messages, use semihosting, it is functional.

closed 2020-08-13 by ilg-ul