xpack-dev-tools/qemu-arm-xpack #7: -daemonize failes with "operation not permitted"

LnnrtS, 2020-11-25:
### Description

emulator does not start with `-daemonize` option

### Steps to Reproduce

works:
```
$ qemu-system-gnuarmeclipse -board STM32F4-Discovery -mcu STM32F407VG -image some/elf/file -S -nographic
QEMU 2.8.0-9 monitor - type 'help' for more information
(qemu)
```

fails:
```
$ qemu-system-gnuarmeclipse -board STM32F4-Discovery -mcu STM32F407VG -image some/elf/file -S -nographic -daemonize
qemu: qemu_mutex_unlock: Operation not permitted
```

### Versions

`2.8.0-9.2`

### Additional information

A quick search bings up [this patch](https://patchwork.kernel.org/project/qemu-devel/patch/1498234919-27316-13-git-send-email-kwolf@redhat.com/)

ilg-ul, 2020-11-25:
Thank you, the current code is quite old, it probably has lots of issues and needs a major update.

I never used `-daemonize`, what is it supposed to do?

LnnrtS, 2020-11-25:
`-daemonize` should make it run in the background.

I was trying to use it for automatic testing from `CTest`.
Without that feature I probably will need some intermediate tool to handle daemonization.

ilg-ul, 2020-11-25:
:-(

I don't know, perhaps it can be easily fixed. You can rebuild qemu and try that patch; if it is functional I can make a new release.